nasa/cFS #417: Where is are the new-app-generation scripts ? (that used to exist in /tools )

obsidian-1, 2022-02-02:
**Checklist (Please check before submitting)**

* [ ] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).
* [ ] I performed a cursory search to see if the bug report is relevant, not redundant, nor in conflict with other tickets.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Code snips**
If applicable, add references to the software.

**System observed on:**
 - Hardware
 - OS: [e.g. Linux 4.4]
 - Versions [e.g. cFE 6.6, OSAL 4.2, PSP 1.3 for mcp750, any related apps]

**Additional context**
Add any other context about the problem here.

**Reporter Info**
Full name and company/organization if applicable

skliper, 2022-02-02:
Good topic for https://github.com/nasa/cFS/discussions.  I'm not aware of anyone who has updated something to work with the latest recommended patterns.  A good starting point is https://github.com/nasa/sample_app or https://github.com/nasa/sample_lib and bulk replace prefixes.

ThibFrgsGmz, 2022-02-12:
I don't know if it follows the latest recommended pattern but you can also grab the script from NASA Icarous: https://github.com/nasa/icarous/tree/master/Python/Template.

At the time I was using it to generate a cFS application.

skliper, 2022-02-14:
Some of these may also help, community provided and I'm not sure how up-to-date they are.  https://github.com/OpenSatKit/OpenSatKit/tree/master/cfs/tools/app_setup_scr, https://github.com/OpenSatKit/OpenSatKit/tree/master/cfs/tools/gen_app_code.

ThibFrgsGmz, 2022-02-14:
@skliper Why not collect them and put them in the bundle? Is the maintenance burden too high?

This need comes up very often.

skliper, 2022-02-14:
@ThibFrgsGmz sample app and lib included the bundle and are intended to provide the starting point for new apps (just global replace SAMPLE_ with your app abbreviation).  Some of the tools I've seen to generate apps basically just do this global replace for you (not huge added value for the additional technical debt of another tool, it's just a `sed`).  Other tools create things from scratch or templates, which ends up duplicating the work of keeping the sample app/lib up to date.  While I understand in certain contexts these tools are very useful (if your organization has sufficient standardized customization that needs to go on top of the samples), those customizations often aren't necessarily generally applicable.  The topic of providing a simple example (sample app/lib) as well as a more complex example exercising a much wider range of the APIs (like creating/managing child tasks, etc) has been discussed but we haven't had the resources to implement.  We could just bake a tool in and use it to create sample app/lib instead of providing the code, but qualification/certifiability/acceptance of another official tool is overhead (and could add dependencies to the basic build).