nasa/cFS #550: format check fails on nasa#ZZZZ

thesamprice, 2022-08-20:
**Checklist (Please check before submitting)**

* [X] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).
* [X] I performed a cursory search to see if the bug report is relevant, not redundant, nor in conflict with other tickets.

**Describe the bug**
See title, on pull requests the commit message is reworded form #zzz to nasa#zzzz

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Code snips**
If applicable, add references to the software.

**System observed on:**
 - Hardware
 - OS: [e.g. Linux 4.4]
 - Versions [e.g. cFE 6.6, OSAL 4.2, PSP 1.3 for mcp750, any related apps]

**Additional context**
Add any other context about the problem here.

**Reporter Info**
Full name and company/organization if applicable

skliper, 2022-08-22:
Can you link to a case where this actually failed?  I haven't seen any issue like this myself, so I suspect it may be more nuanced.  And/or submit example PRs that show this rejection behavior?

thesamprice, 2022-08-22:
I think it was because i had 'fix' in the title rather than Fix.
It might be wroth adding git hooks for message formats to the repos that are enforcing this.

I think it would look like this

cat .githooks/commit-msg
```bash
#!/bin/bash
#
# Purpose: Check that[NAV-XYZ] is prepended to the commit message

# Check for issue tag in commit message
issue_tag_regex="^((Fix|HotFix|Part)\s\#[0-9]+,\s[a-zA-Z0-9]+|Merge\spull\srequest\s\#[0-9]+\s[a-zA-Z0-9]+|IC:\s[a-zA-Z0-9]+)"

if ! grep -qE "$issue_tag_regex" "$1"; then

    cat << EOF
ERROR - Commit message does not contain proper format
Expected Commit Format: (Fix|HotFix|Part) #ZZZ, message
EOF

    exit 1
fi

exit 0
```