nasa/cFS #560: Remove rules in CodeQL coding standard

ArielSAdamsNASA, 2022-09-01:
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).
* [x] I reviewed the [cFS README.md file](https://github.com/nasa/cFS/blob/main/README.md) to see if the feature is in the major future work.
* [x] I performed a cursory search to see if the feature request is relevant, not redundant, nor in conflict with other tickets.

**Is your feature request related to a problem? Please describe.**
The CodeQL (coding standard) workflow fails because there are too many results to be uploaded. 

**Describe the solution you'd like**
Remove the rules that produce the most results until we resolve or dismiss the other results. 

**Describe alternatives you've considered**
If we want to show results for all coding standard rules, then we will need multiple matrixes. There can be one for all MISRA rules and JPL rules except 14, 17, and 24. JPL rules 14, 17, and 24 would have their own job/matrix. 

**Additional context**
JPL rules 14, 17, and 24 each have over 5,000 results which is the limit. Without rules 14, 17, and 24, the results are uploaded, however, 2,394 rules are ignored (https://github.com/ArielSAdams/cFS/security/code-scanning?page=200&query=is%3Aopen+branch%3ANew-CodeQL-Config). Therefore, JPL 15 (2,920 results) would also need to be ignored or have its own matrix. 

**Requester Info**
Ariel Adams, MCSG Technologies

skliper, 2022-09-01:
I think for now suppressing at least 14 (checking return value), 17 (explicitly sized types) for now would be good.  I'm also OK w/ 15 (checking parameter validity) suppressing for now, although some day we may want to re-enable or consider checking at least APIs.  I'm a little surprised about 24 (multiple statements or declarations on one line)... do you have an example?  I haven't seen this warning that frequently using other analyzers, maybe there's a specific pattern it is picking up?

ArielSAdamsNASA, 2022-09-01:
@skliper Here is an [example](https://github.com/ArielSAdams/cFS/security/code-scanning/29147) from this list: https://github.com/ArielSAdams/cFS/security/code-scanning?query=is%3Aopen+branch%3AJPL-24. Unfortunately, CodeQL still has that issue where it doesn't show code previews for submodules/subrepositories. I can run JPL 24 by itself in OSAL for a code preview if needed.

skliper, 2022-09-01:
@ArielSAdamsNASA - thanks for submitting [CodeQL - false positive - JPL Rule 24 · Issue #6522 · github/codeql](https://github.com/github/codeql/issues/6522)!  I posted a comment/example here https://github.com/github/codeql/issues/6522#issuecomment-1234565381.  Until there's a better way to address it I'm fine with suppressing 24 (note it's still enabled in our internal static analysis as a style warning, which only shows up a handful of places).

commit 4d744cf5 (2022-09-01): Fix #560, Remove Rules in CodeQL Coding Standard
commit b5d5bb22 (2022-09-01): Merge pull request #561 from ArielSAdams/cFS:fix-560-remove-rules-codeql

Fix #560, Remove Rules in CodeQL Coding Standard
PR #561 (2022-09-01): Fix #560, Remove Rules in CodeQL Coding Standard
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).
* [ ] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
Fixes #560. Removes JPL rules 14, 15, 17, and 24. 

**Testing performed**
Tested on fork

**Expected behavior changes**
CodeQL workfl [... cut]
closed 2022-09-01 by dzbaker