nasa/cFS #566: Automate generation of CCB Agenda

ArielSAdams, 2022-09-07:
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).
* [x] I reviewed the [cFS README.md file](https://github.com/nasa/cFS/blob/main/README.md) to see if the feature is in the major future work.
* [x] I performed a cursory search to see if the feature request is relevant, not redundant, nor in conflict with other tickets.

**Is your feature request related to a problem? Please describe.**
Reduce time spent on manually creating CCB Agenda. 

**Describe the solution you'd like**
Create a new workflow that uses the workflow_dispatch event. The workflow should identify PRs and their issues that have the label CCB Ready. The document should list the author, PR title, link to PR, and the PR linked issue title. This file should be uploaded to the Wiki. 

**Describe alternatives you've considered**
Keep as is. 

**Additional context**
Will most likely need to create new action.
Possible actions that may be used: 
1. https://github.com/marketplace/actions/list-pull-requests
- Outputs PR title and link based on label. Doesn't work across multiple repos. 
2. https://github.com/budziq/pulls_since
- Outputs PR author, PR title, and link. Works for any repo/multiple repos. Doesn't pull based on labels. 
3. https://github.com/sshaw/export-pull-requests
- Outputs PR author, PR title, PR link, and issue based on label. Creates CSV file. Needs to convert to a markdown file for Wiki repo. 

Github repo wiki is actually a repo with named reponame.wiki. 

**Requester Info**
Ariel Adams, MCSG Tech.

skliper, 2022-09-07:
I recommend considering adding every PR that doesn't have the CCB:Ignore label, or come up with a process to assign PR's without the CCB:Ready/Ignore labels to cFS team members to review and promote to CCB:Ready... otherwise community PR's get neglected and/or PR's stagnate.  Note the CCB:Ignore label should be accompanied by a comment as to what the path forward is.

skliper, 2022-09-07:
Note you could also probably automate via one of the GitHub CLI's, either hub or gh could likely generate the list across all repos w/ a fairly simple script.

skliper, 2022-09-07:
List of all non CCB:Ignore PRs (over all repos) sorted by user:
```
(repolist="CF CS DS FM HK HS LC MD MM SC cFE osal PSP sample_app sample_lib ci_lab to_lab sch_lab cFS-GroundSystem elf2cfetbl tblCRCTool"; for repo in $repolist; do   gh api --method GET -H "Accept: application/vnd.github.v3+json" /repos/nasa/$repo/pulls --jq '.[] | select(any(.labels[].name ; contains("CCB:Ignore")) | not) | "\(.user.login): \(.url) \(.title)"'; done) | sort
```

commit 26193a92 (2022-11-17): Fix #566, Adds CB Agenda generation workflow
commit fb37be94 (2022-11-17): Merge pull request #598 from chillfig/ccbAgenda

Fix #566, Adds CCB Agenda generation workflow
PR #598 (2022-11-17): Fix #566, Adds CCB Agenda generation workflow
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
- Fixes #566 

**Testing performed**
Test on fork: https://github.com/chillfig/cFS/actions/runs/3338091197
1) Navigate to "update wiki"  [... cut]
closed 2022-11-17 by dzbaker