nasa/cFS #639: cFS Draco Release

Guitarum, 2023-01-06:
Hello,

We are stating a project based on the cFS framework for my company.

We noticed that the latest official release is CFE 6.7, and since then only "release candidates" have been released. The latest is draco-rc3 (for 2022), and we would like to use this version as our baseline to start our development.

The questions are:
- Will draco become an official release later on? If not, is there a plan for the next official release?
- Is there any reasons why since CFE 6.7 only release candidates have been published? All the documentation seems adequate (Requirements, traceabilities, unit tests and code coverage results, release notes) for each cFS core and app components, and draco-rc3 seems fully validated.

We simply wanted to confirm this point before we go ahead in using draco.

Many thanks in advance,

Guillaume

ivanperez-keera, 2023-04-11:
Also interested.

rlourette, 2023-05-20:
also interested

Guitarum, 2023-07-15:
Hi all,

I am wondering if you have any update on this? A draco-rc4 pre-release is now available, and a equuleus-rc1 is in progress. Do you know is there is an official release planned on the cFS roadmap?

Many thanks,

Guillaume

ivanperez-keera, 2023-11-27:
I think part of the issue is that the releases have a note saying that they are pre-releases and nobody edits them once a release is final / frozen. Maybe that's all that'd be needed.