nasa/cFS #645: Update format check workflow to newer clang-format version

jphickey, 2023-01-19:
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).x
* [x] I reviewed the [cFS README.md file](https://github.com/nasa/cFS/blob/main/README.md) to see if the feature is in the major future work.
* [x] I performed a cursory search to see if the feature request is relevant, not redundant, nor in conflict with other tickets.

**Is your feature request related to a problem? Please describe.**
The format check workflow is currently running on ubuntu 20.04, which in turn includes clang-format version 10 by default.

Versions of clang-format up through 12 have a bug/limitation where the `PointerAlignment` setting is not implemented.  The cFS format config has this set as "Right":
https://github.com/nasa/cFS/blob/7f11bdfd1b80a42c1058375f2f992ca63c96e56d/.clang-format#L50

However, in version 13 this was fixed, and it now honors this setting.  As such, if a user has a newer version, they will get format discrepancies when trying to submit a PR.  This is the case when using Ubuntu 22.04 as it includes version 14 by default.   For an example, see any struct that contains a pointer, such as:

`void *             Addr`

now needs to be:

`void              *Addr` 

due to the `PointerAlignment: Right` now taking effect.  Unfortunately, setting `PointerAlignment: Left` does not replicate the version 10 format either, it produces this instead:

`void*              Addr`

Sadly it seems there is no backward-compatible option to replicate the version 10 format with version 14 (or at least none that I could find).   Furthermore, it is not possible to install version 10 in Ubuntu 22.04 either.

**Describe the solution you'd like**
Eventually, we will have to upgrade to a newer version of clang-format, and deal with these discrepancies.  As a transition, version 14 is easily installable on Ubuntu 20.04 via the packages offered from `llvm.org` (official upstream for clang).  

Recommendation would be to change the format-check workflow to run version 14 instead of version 10, and update the code accordingly for all of the whitespace differences it produces similar to the above example.  This can be done by simply bumping the `runs-on` to be ubuntu 22.04 instead of 20.04, and changing nothing else (note that this needs to happen eventually anyway, since `ubuntu-latest` is now 22.04 on github, and 20.04 will be deprecated at some point, although this may be a while yet).

**Describe alternatives you've considered**
In the interim, it is possible to install clang-format version 12 on both Ubuntu 20.04 and 22.04, which seems to produce the same format as v10 did (at least with cFS config file).  But this is not the default version in either, and only defers the issue, because v12 depends on a version of python that is no longer available in 22.10 or newer and thus becomes non-installable in anything newer.  So this does not resolve the incompatibility, it only adds setup complexity while delaying the eventuality of having to deal with the whitespace changes.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

skliper, 2023-01-19:
I agree.  We were upstream when we used 10 with Ubuntu 18.04, no concerns w/ upstream 14 on Ubuntu 20.  Plus I like the 14 format way better, so I'm all in.

skliper, 2023-01-19:
Actually, I didn't read this closely enough.  Ubuntu 22.04 is not yet approved for use by us internally, so instead of switching run-on to 22, I'd rather it did 20 with upstream 14.  Not a big deal, but without internal approval to use 22 we can't really replicate 22 workflows locally.