nasa/cFS #656: OSAL workflow "Static Analysis" fails to convert cppcheck results to SARIF

frank-kue, 2023-03-09:
* [x] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).
* [x] I performed a cursory search to see if the bug report is relevant, not redundant, nor in conflict with other tickets.

**Describe the bug**
When running the *Static Analysis* workflow on a clone of the OSAL repository, the workflow fails at the step *Convert cppcheck results to SARIF* since about 2023-03-08 evening MET. One day before I successfully executed the same workflow without any error. The error is:

~~~
Run npx "@microsoft/sarif-multitool" convert "cppcheck_err.xml" --tool "CppCheck" --output "cppcheck_err.sarif" --force
  npx "@microsoft/sarif-multitool" convert "cppcheck_err.xml" --tool "CppCheck" --output "cppcheck_err.sarif" --force
  shell: /usr/bin/bash --noprofile --norc -e -o pipefail {0}
  env:
    CPPCHECK_OPTS: --project=/home/runner/work/osal/osal/build/compile_commands.json
SARIF Multitool 4.1.0
© Microsoft Corporation. All rights reserved.

ERROR(S):
  Option 'force' is unknown.
[...]
~~~

I am new to Github workflows. Therefore, this could easily be caused by a mistake on my side. Yet, at the first glance the error messages does not look like I am guilty. On the other hand, I cannot imagine someone removed the `--force` option from a tool overnight. Maybe you can hint me towards my mistake.

I am not sure whether the links below will work for you:

* [This was a successful execution of the workflow 2023-03-07 (already with `--force` flag)](https://github.com/frank-kue/osal/actions/runs/4353883132/jobs/7613984260)
* [This is a failure of a re-run from today (2023-03-09)](https://github.com/frank-kue/osal/actions/runs/4353883132/jobs/7655733695)

**To Reproduce**
Steps to reproduce the behavior:

1. Create a clone of the OSAL repository, in the *Actions* tab enable workflows, in the *Settings* tab → open *Actions* (in the left column) → *General* → in *Workflow permissions* → tick *Read write permission* → *Save*
2. In the *Actions* tab, click on the *Static Analysis* workflow, from *Run workflow* select *Branch: main*, then click on *Run workflow*
3. Wait till the new workflow appears in the list and terminates execution (takes about one minute)
4. Click on the workflow entry in the list (the new workflow will have a red bubble in front of it)
5. In the box *Annotations* click on *Run Static Analysis*
6. A black scrolling widget with the workflow steps and their output will appear.
7. Scroll up to the place with the red bubble on the left side, titled *Convert cppcheck results to SARIF*
8. See the above error message just below of it

**Expected behavior**
The *Static Analysis* workflow should execute without errors in the steps which convert the results in other formats. 

**Code snips**
-/-

**System observed on:**
Github, clone of OSAL repository

**Additional context**
-/-

**Reporter Info**
Frank Kühndel – embedded-brains GmbH

jphickey, 2023-03-09:
It looks like there was some recent change in the upstream sarif conversion tool from Microsoft, this appears to be showing up in all the workflows suddenly

frank-kue, 2023-03-13:
@jphickey @dzbaker Many thanks!

commit b6d7e268 (2023-03-13): Fix #656, do not use --force option on sarif conversion

Looks like this option was removed in a recent update
commit f28bab24 (2023-03-13): Merge pull request #657 from jphickey/fix-656-sarif-noforce

Fix #656, do not use --force option on sarif conversion