nasa/cFS #672: portable doesn't mean MacOS

deepgray, 2023-05-10:
**Checklist (Please check before submitting)**

* [ ] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).
* [ ] I performed a cursory search to see if the bug report is relevant, not redundant, nor in conflict with other tickets.

**Describe the bug**
trying to build on MacOS, the build fails as no Darwin support.

**To Reproduce**
Steps to reproduce the behavior:
try building on a MacOS platform.

**Expected behavior**
portable i would take as any platform.

**Code snips**
If applicable, add references to the software.

**System observed on:**
 Mac Pro (2013) intel xenon processor
MacOS 12.6.5

**Additional context**
Add any other context about the problem here.

**Reporter Info**
Full name and company/organization if applicable

deepgray, 2023-05-10:
ex0atmospheric@Dad-Mac-Pro cFS % make SIMULATION=native prep
mkdir -p "build"
(cd "build" && cmake  -DCMAKE_INSTALL_PREFIX=/exe -DCMAKE_BUILD_TYPE=debug "/Users/ex0atmospheric/Projects/cFS/cfe")
-- OMIT_DEPRECATED=false: Deprecated elements included in build
-- Search path for modules: apps;libs;psp/fsw/modules;cfe/modules
-- Module 'core_api' found at /Users/ex0atmospheric/Projects/cFS/cfe/modules/core_api
-- Module 'core_private' found at /Users/ex0atmospheric/Projects/cFS/cfe/modules/core_private
-- Module 'ci_lab' found at /Users/ex0atmospheric/Projects/cFS/apps/ci_lab
-- Module 'to_lab' found at /Users/ex0atmospheric/Projects/cFS/apps/to_lab
-- Module 'sch_lab' found at /Users/ex0atmospheric/Projects/cFS/apps/sch_lab
-- Module 'cfe_assert' found at /Users/ex0atmospheric/Projects/cFS/cfe/modules/cfe_assert
-- Module 'sample_app' found at /Users/ex0atmospheric/Projects/cFS/apps/sample_app
-- Module 'sample_lib' found at /Users/ex0atmospheric/Projects/cFS/libs/sample_lib
-- Module 'es' found at /Users/ex0atmospheric/Projects/cFS/cfe/modules/es
-- Module 'evs' found at /Users/ex0atmospheric/Projects/cFS/cfe/modules/evs
-- Module 'fs' found at /Users/ex0atmospheric/Projects/cFS/cfe/modules/fs
-- Module 'sb' found at /Users/ex0atmospheric/Projects/cFS/cfe/modules/sb
-- Module 'tbl' found at /Users/ex0atmospheric/Projects/cFS/cfe/modules/tbl
-- Module 'time' found at /Users/ex0atmospheric/Projects/cFS/cfe/modules/time
-- Module 'osal' found at /Users/ex0atmospheric/Projects/cFS/osal
-- Module 'psp' found at /Users/ex0atmospheric/Projects/cFS/psp
-- Module 'msg' found at /Users/ex0atmospheric/Projects/cFS/cfe/modules/msg
-- Module 'sbr' found at /Users/ex0atmospheric/Projects/cFS/cfe/modules/sbr
-- Module 'resourceid' found at /Users/ex0atmospheric/Projects/cFS/cfe/modules/resourceid
-- Module 'config' found at /Users/ex0atmospheric/Projects/cFS/cfe/modules/config
-- OSAL_SYSTEM_BSPTYPE not defined, no OSAL implementation will be compiled
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/core_api/config/default_cfe_mission_cfg.h for mission_cfg.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/sample_defs/sample_perfids.h for perfids.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/core_api/config/default_cfe_core_api_interface_cfg.h for cfe_core_api_interface_cfg.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/es/config/default_cfe_es_mission_cfg.h for cfe_es_mission_cfg.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/es/config/default_cfe_es_interface_cfg.h for cfe_es_interface_cfg.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/es/config/default_cfe_es_extern_typedefs.h for cfe_es_extern_typedefs.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/es/config/default_cfe_es_fcncodes.h for cfe_es_fcncodes.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/es/config/default_cfe_es_msgdefs.h for cfe_es_msgdefs.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/es/config/default_cfe_es_msg.h for cfe_es_msg.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/es/config/default_cfe_es_msgstruct.h for cfe_es_msgstruct.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/es/config/default_cfe_es_topicids.h for cfe_es_topicids.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/evs/config/default_cfe_evs_mission_cfg.h for cfe_evs_mission_cfg.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/evs/config/default_cfe_evs_interface_cfg.h for cfe_evs_interface_cfg.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/evs/config/default_cfe_evs_extern_typedefs.h for cfe_evs_extern_typedefs.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/evs/config/default_cfe_evs_fcncodes.h for cfe_evs_fcncodes.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/evs/config/default_cfe_evs_msgdefs.h for cfe_evs_msgdefs.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/evs/config/default_cfe_evs_msg.h for cfe_evs_msg.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/evs/config/default_cfe_evs_msgstruct.h for cfe_evs_msgstruct.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/evs/config/default_cfe_evs_topicids.h for cfe_evs_topicids.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/fs/config/default_cfe_fs_mission_cfg.h for cfe_fs_mission_cfg.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/fs/config/default_cfe_fs_interface_cfg.h for cfe_fs_interface_cfg.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/fs/config/default_cfe_fs_filedef.h for cfe_fs_filedef.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/fs/config/default_cfe_fs_extern_typedefs.h for cfe_fs_extern_typedefs.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/sb/config/default_cfe_sb_mission_cfg.h for cfe_sb_mission_cfg.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/sb/config/default_cfe_sb_interface_cfg.h for cfe_sb_interface_cfg.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/sb/config/default_cfe_sb_extern_typedefs.h for cfe_sb_extern_typedefs.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/sb/config/default_cfe_sb_fcncodes.h for cfe_sb_fcncodes.h
-- Using file: /Users/ex0atmospheric/Projects/cFS/cfe/modules/sb/config/default_cfe_sb_msgdefs.h for cfe_sb_msgdefs.h
[... cut ...]
CMake Error at cmake/arch_build.cmake:648 (message):
  Do not know how to set CFE_SYSTEM_PSPNAME on Darwin system
Call Stack (most recent call first):
  CMakeLists.txt:120 (prepare)

-- Configuring incomplete, errors occurred!
CMake Error at cmake/mission_build.cmake:492 (message):
  Failed to configure native_default_cpu1
Call Stack (most recent call first):
  CMakeLists.txt:124 (process_arch)

-- Configuring incomplete, errors occurred!
make: *** [prep] Error 1
ex0atmospheric@Dad-Mac-Pro cFS %

avan989, 2023-08-10:
I think it would be better to use a VM or docker with one of the supported OS. But from just looking at the error message, it looks like you are missing the following from whatever "toolchain" file you are using to compile.  

```
SET(CFE_SYSTEM_PSPNAME      ...)
SET(OSAL_SYSTEM_BSPTYPE     ...)
SET(OSAL_SYSTEM_OSTYPE      ...)
```

muxeh, 2025-01-04:
Was this ever resolved? I'm trying to build on MacOS and running into the same problems

dzbaker, 2025-01-06:
> Was this ever resolved? I'm trying to build on MacOS and running into the same problems

Hi @muxeh, this one hasn't been resolved. We don't officially support MacOS at the moment, although I would recommend using a VM or container with one of the supported OSes. Let me know if this helps.

closed 2025-03-04 by dzbaker