nasa/cFS #709: Renaming the project name to the name of the containing folder causes make SIMULATION=native prep to fail

embeddedpenguin, 2023-10-04:
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).
* [x] I performed a cursory search to see if the bug report is relevant, not redundant, nor in conflict with other tickets.

**Describe the bug**
When creating a new cfs project, if the name of the project folder is the same name as the _defs folder, the `make SIMULATION=native prep` fails.

**To Reproduce**
Steps to reproduce the behavior:
Create a new cfs project (git clone this repo)
```
git clone <this repo> testcfs
cp -r cfe/cmake/sample_defs testcfs_defs
cp cfe/cmake/Makefile.sample Makefile
cd example_defs; rename 's/example/testcfs/' example_*; mv sample_perfids.h testcfs_perfids.h; cd ..
make distclean
make SIMULATION=native prep
```
**Actual behavior**
<details>
<summary>build output</summary>
mkdir -p "build"
(cd "build" && cmake -DCMAKE_INSTALL_PREFIX=/exe -DCMAKE_BUILD_TYPE=debug "/home/user/Projects/testcfs/cfe")
-- OMIT_DEPRECATED=false: Deprecated elements included in build
-- Search path for modules: apps;libs;psp/fsw/modules;cfe/modules
-- Module 'core_api' found at /home/user/Projects/testcfs/cfe/modules/core_api
-- Module 'core_private' found at /home/user/Projects/testcfs/cfe/modules/core_private
-- Module 'ci_lab' found at /home/user/Projects/testcfs/apps/ci_lab
-- Module 'to_lab' found at /home/user/Projects/testcfs/apps/to_lab
-- Module 'sch_lab' found at /home/user/Projects/testcfs/apps/sch_lab
-- Module 'cfe_assert' found at /home/user/Projects/testcfs/cfe/modules/cfe_assert
-- Module 'sample_app' found at /home/user/Projects/testcfs/apps/sample_app
-- Module 'sample_lib' found at /home/user/Projects/testcfs/libs/sample_lib
-- Module 'es' found at /home/user/Projects/testcfs/cfe/modules/es
-- Module 'evs' found at /home/user/Projects/testcfs/cfe/modules/evs
-- Module 'fs' found at /home/user/Projects/testcfs/cfe/modules/fs
-- Module 'sb' found at /home/user/Projects/testcfs/cfe/modules/sb
-- Module 'tbl' found at /home/user/Projects/testcfs/cfe/modules/tbl
-- Module 'time' found at /home/user/Projects/testcfs/cfe/modules/time
-- Module 'osal' found at /home/user/Projects/testcfs/osal
-- Module 'psp' found at /home/user/Projects/testcfs/psp
-- Module 'msg' found at /home/user/Projects/testcfs/cfe/modules/msg
-- Module 'sbr' found at /home/user/Projects/testcfs/cfe/modules/sbr
-- Module 'resourceid' found at /home/user/Projects/testcfs/cfe/modules/resourceid
-- Module 'config' found at /home/user/Projects/testcfs/cfe/modules/config
-- OSAL_SYSTEM_BSPTYPE not defined, no OSAL implementation will be compiled
-- Using file: /home/user/Projects/testcfs/cfe/modules/core_api/config/default_cfe_mission_cfg.h for mission_cfg.h
CMake Error at cmake/global_functions.cmake:83 (message):
  No implementation for perfids.h found
Call Stack (most recent call first):
  cmake/global_functions.cmake:186 (cfe_locate_implementation_file)
  modules/core_api/mission_build.cmake:20 (generate_config_includefile)
  cmake/mission_build.cmake:356 (include)
  CMakeLists.txt:120 (prepare)

-- Configuring incomplete, errors occurred!
See also "/home/user/Projects/testcfs/build/CMakeFiles/CMakeOutput.log".
make: *** [Makefile:126: prep] Error 1

</details>
<details>
<summary>build/CMakeFiles/CMakeOutput.log</summary>
The system is: Linux - 6.2.0-33-generic - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/gcc-12 
Build flags: 
Id flags:  

The output was:
[... 218 lines cut ...]
├── default_osconfig.cmake
├── global_build_options.cmake
├── mission_build_custom.cmake
├── native_osconfig.cmake
├── targets.cmake
├── testcfs_mission_cfg.h
├── testcfs_perfids.h
├── testcfs_platform_cfg.h
├── toolchain-arm-cortexa8_neon-linux-gnueabi.cmake
├── toolchain-i686-linux-clang.cmake
├── toolchain-i686-linux-gnu.cmake
├── toolchain-i686-rtems4.11.cmake
├── toolchain-i686-rtems5.cmake
├── toolchain-i686-rtems6.cmake
├── toolchain-powerpc-440_softfp-linux-gnu.cmake
└── toolchain-ppc-vxworks6.9.cmake
```

**Expected behavior**
The project shouldn't care what the parent directory is named. It should build successfully.
If I change the directory name to anything else, like penguinswearingsunglasses, it runs the prep target fine.

**System observed on:**
- Linux 6.2.0
- Ubuntu 22.04

skliper, 2024-01-05:
It's not finding `testcfs_perfids.h`.  If you prep with `VERBOSE=1` it'll print valid names/locations for this file.

skliper, 2024-01-05:
Note the setup steps fail as written (cd example_defs, but you copy it as testcfs_defs).  It doesn't look like your build output is from what's described, since testcfs_perfids.h should be a valid name in this case.  Can you recreate and confirm there's a real issue, and/or provide setup steps that can recreate the issue?  The handful of lines output with `VERBOSE=1` where it searches for the perfids.h file would also help if there's still an issue.

embeddedpenguin, 2024-01-06:
The reason why the original instructions didn't work is because I'm using a cfs project generation tool for my purposes, and I tried to change the instructions from what I was using for that project to match with what you would expect. I've confirmed the instructions below should work for anyone.

I've fixed the instructions for reproducing:
NOTE: This is using the rename utility provided by the util-linux package, _not_ the rename package. The original instructions were using the rename package.
```
git clone https://github.com/nasa/cFS.git testcfs
cd testcfs
git submodule init
git submodule update --recursive
cp -r cfe/cmake/sample_defs testcfs_defs
cp cfe/cmake/Makefile.sample Makefile
cd testcfs_defs; rename example testcfs example_*; mv sample_perfids.h testcfs_perfids.h; cd ..
# sanity purge
make distclean
VERBOSE=1 make SIMULATION=native prep
```

The above fails with the following output:
```
mkdir -p "build"
(cd "build" && cmake -DCMAKE_INSTALL_PREFIX=/exe -DCMAKE_BUILD_TYPE=debug "/home/user/Projects/testcfs/cfe")
-- The C compiler identification is GNU 13.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Mission configuration testcfs automatically selected
-- OMIT_DEPRECATED=false: Deprecated elements included in build
-- Search path for modules: apps;libs;psp/fsw/modules;cfe/modules
-- Module 'core_api' found at /home/user/Projects/testcfs/cfe/modules/core_api
-- Module 'core_private' found at /home/user/Projects/testcfs/cfe/modules/core_private
-- Module 'ci_lab' found at /home/user/Projects/testcfs/apps/ci_lab
-- Module 'to_lab' found at /home/user/Projects/testcfs/apps/to_lab
-- Module 'sch_lab' found at /home/user/Projects/testcfs/apps/sch_lab
-- Module 'cfe_assert' found at /home/user/Projects/testcfs/cfe/modules/cfe_assert
-- Module 'sample_app' found at /home/user/Projects/testcfs/apps/sample_app
-- Module 'sample_lib' found at /home/user/Projects/testcfs/libs/sample_lib
-- Module 'es' found at /home/user/Projects/testcfs/cfe/modules/es
-- Module 'evs' found at /home/user/Projects/testcfs/cfe/modules/evs
-- Module 'fs' found at /home/user/Projects/testcfs/cfe/modules/fs
-- Module 'sb' found at /home/user/Projects/testcfs/cfe/modules/sb
-- Module 'tbl' found at /home/user/Projects/testcfs/cfe/modules/tbl
-- Module 'time' found at /home/user/Projects/testcfs/cfe/modules/time
-- Module 'osal' found at /home/user/Projects/testcfs/osal
-- Module 'psp' found at /home/user/Projects/testcfs/psp
-- Module 'msg' found at /home/user/Projects/testcfs/cfe/modules/msg
-- Module 'sbr' found at /home/user/Projects/testcfs/cfe/modules/sbr
-- Module 'resourceid' found at /home/user/Projects/testcfs/cfe/modules/resourceid
-- Module 'config' found at /home/user/Projects/testcfs/cfe/modules/config
-- OSAL_SYSTEM_BSPTYPE not defined, no OSAL implementation will be compiled
-- Using file: /home/user/Projects/testcfs/cfe/modules/core_api/config/default_cfe_core_api_interface_cfg.h for core_api_interface_cfg.h
-- Using file: /home/user/Projects/testcfs/cfe/modules/core_api/config/default_cfe_mission_cfg.h for mission_cfg.h
CMake Error at cmake/global_functions.cmake:83 (message):
  No implementation for perfids.h found
Call Stack (most recent call first):
  cmake/global_functions.cmake:186 (cfe_locate_implementation_file)
  modules/core_api/mission_build.cmake:43 (generate_config_includefile)
  cmake/mission_build.cmake:431 (include)
  CMakeLists.txt:120 (prepare)

-- Configuring incomplete, errors occurred!
make: *** [Makefile:126: prep] Error 1
```

Immediately after that output, I can bandaid fix the project to prep properly via the following instructions:
```
cd ..
# rename to anything other than testcfs
mv testcfs testcfszzz
[... 182 lines cut ...]
-- Building Core Module: sbr
-- Using direct map software bus routing implementation
-- Building Core Module: resourceid
-- Building Core Module: config
-- Building Dynamic App: ci_lab targets=cpu1
-- Building Dynamic App: to_lab targets=cpu1
-- Using file: /home/user/Projects/testcfszzz/apps/to_lab/fsw/tables/to_lab_sub.c for to_lab_sub.c
-- Using /home/user/Projects/testcfszzz/apps/to_lab/fsw/tables/to_lab_sub.c as table definition for to_lab_sub on cpu1
-- Building Dynamic App: sch_lab targets=cpu1
-- Using file: /home/user/Projects/testcfszzz/apps/sch_lab/fsw/tables/sch_lab_table.c for sch_lab_table.c
-- Using /home/user/Projects/testcfszzz/apps/sch_lab/fsw/tables/sch_lab_table.c as table definition for sch_lab_table on cpu1
-- Building Dynamic App: cfe_assert targets=cpu1
-- Building Dynamic App: sample_app targets=cpu1
-- Using file: /home/user/Projects/testcfszzz/apps/sample_app/fsw/tables/sample_app_tbl.c for sample_app_tbl.c
-- Using /home/user/Projects/testcfszzz/apps/sample_app/fsw/tables/sample_app_tbl.c as table definition for sample_app_tbl on cpu1
-- Building Dynamic App: sample_lib targets=cpu1
NOTE: Selected /home/user/Projects/testcfszzz/testcfs_defs/cpu1_cfe_es_startup.scr as source for cfe_es_startup.scr on cpu1
-- Configuring done (0.1s)
-- Generating done (0.1s)
-- Build files have been written to: /home/user/Projects/testcfszzz/build/native/default_cpu1
-- Configuring done (0.3s)
-- Generating done (0.0s)
-- Build files have been written to: /home/user/Projects/testcfszzz/build
echo "-DCMAKE_INSTALL_PREFIX=/exe -DCMAKE_BUILD_TYPE=debug" > "build/.prep"
```

embeddedpenguin, 2024-01-06:
I'll try to find some time this weekend to look into it, but the starting point is
from failing output:
```
CMake Error at cmake/global_functions.cmake:83 (message):
  No implementation for perfids.h found
```

vs the successful output:
```
-- Using file: /home/user/Projects/testcfszzz/testcfs_defs/testcfs_perfids.h for perfids.h
```

Doesn't seem like cmake is filling ______perfids.h with testcfs when the parent directory is named testcfs.

skliper, 2024-01-08:
I was able to recreate, although not exactly following your example  since I'm still a bit confused wrt your setup.  It doesn't look like util-linux provides rename on Ubuntu 22.04.  Anyways, it comes down to this logic:
https://github.com/nasa/cFE/blob/7f5ebcd15032fe31e7ef0e5ff4ef624d51d66e16/cmake/global_functions.cmake#L57-L63

It should probably only check for a match relative to MISSION_DEFS directory instead of the full path.  Or just remove the test, I'm not sure why it needs to be that strict (could still test for something w/ a prefix that matches the relative dir).

skliper, 2024-01-08:
I got the change ready, I'll tag you on it :)

embeddedpenguin, 2024-01-29:
Fixed via https://github.com/nasa/cFE/pull/2494

closed 2024-01-29 by embeddedpenguin