nasa/cFS #743: 79 - coverage-es-ALL (SEGFAULT)

embeddedpenguin, 2024-02-17:
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).
* [x] I performed a cursory search to see if the bug report is relevant, not redundant, nor in conflict with other tickets.

**Describe the bug**
Test fixture 79 segfaults, and I can't figure out if it's intended behavior. (I hope not!)

**To Reproduce**
Steps to reproduce the behavior:
```
git clone https://github.com/nasa/cFS.git --recurse-submodules
cd cFS
SIMULATION=native ENABLE_UNIT_TESTS=1 make prep
make -j$(nproc)
cd build/native/default_cpu1
ctest -j$(nproc) # this is just to speed up the process
ctest --rerun-failed --output-on-failure

**Expected behavior**
It looks like the specific test, found [here](https://github.com/nasa/cFE/blob/8cdad66e4eafd11c2ea93d0360867045b77fa981/modules/es/ut-coverage/es_UT.c#L4557) mentions that it's forcing a vsnprintf failure, so I expect a failure that would lead to the test passing.

**Actual behavior**
A segfault occurs, causing the test fixture to fail.

**System observed on:**
- OS/Hardware: 
Linux gpenguin 6.6.13-gentoo-dist #1 SMP PREEMPT_DYNAMIC Fri Jan 26 12:11:22 CST 2024 x86_64 AMD Ryzen 9 5950X 16-Core Processor AuthenticAMD GNU/Linux
 - Versions: N/A (directly from main)

**Additional context**
This might not be a bug. Is this test expected to segfault? If it is, any advice on how I should handle this in CI? 

Attached is a log containing the output of the failed unit test.

[cover-es-ALL-segfault.log](https://github.com/nasa/cFS/files/14317845/cover-es-ALL-segfault.log)

embeddedpenguin, 2024-02-17:
After debugging this further, I'm not sure what this test is supposed to be doing. It says it's supposed to force a failure by supplying NULL as the fmt option, but why? Is it just to make

https://github.com/nasa/cFE/blob/8cdad66e4eafd11c2ea93d0360867045b77fa981/modules/es/ut-coverage/es_UT.c#L4559
Shouldn't this always segfault? Is the test trying to make sure that it segfaults?

embeddedpenguin, 2024-02-19:
I now understand what the test is doing, but I can't seem to catch the segfault.

closed 2024-05-27 by embeddedpenguin

MichaelBonnet, 2025-03-12:
I have noted that this same unit test segfaults on local for me, but for some reason never does in CI.