nasa/cFS #766: How hardware components are integrated with cFS applications?

Gonta01, 2024-05-15:
There are [CFS-101 TRAINING GUIDE](https://software.nasa.gov/software/MSC-26323-1) where in details is described how to create own application. The examples are based on temperature monitoring.

The issue is how temperature is transmitted from hardware layer up to applications for real board? What should be added and at what layer? Should PSP, BSP layers be somehow modified?
The issue is not only related to temperature but any interaction with any hardware isn't clear (at least for me).

Are there any examples, guide or documents that could clarify this issue?

open-stemware, 2024-05-15:
I couldn't have asked for a better lead in...  There are two cFS Basecamp projects that provide working examples for what was just described.

[Raspberry Pi LED Control](https://openmissionstack.com/projects_read/gpio_demo) is an example of a hardware interface library serving as a device driver for an app that controls an LED. 

[Payload Manager](https://openmissionstack.com/projects_read/payload_manager) is an example of simulating a hardware device when the hardware is not available, or a hardware simulator is not available for testing error paths.

Gonta01, 2024-05-15:
@skliper 

> See sample_app/lib for makefile examples.

Do you mean [sample_lib](https://github.com/nasa/sample_lib/tree/83d26957beaf256bbcdff3188a67dc4025726704) or [sample_app](https://github.com/nasa/sample_app/tree/ceaf7ab5fadc7ac80f9bee6f5a2757593417e461)?
I haven't understood to what exactly you are referring.

skliper, 2024-05-15:
@Gonta01 - Yes, for CMake example of an app calling a lib and how to add them to cFS.  Although it sounds like @open-stemware has examples more specific to your use case.

open-stemware, 2024-05-15:
In case you don't want to work through the Basecamp projects here are links to the lib/app repos used in each project:

**Raspberry Pi LED Control**
[PI IO Lib](https://github.com/cfs-apps/pi_iolib)
[LED Control App](https://github.com/cfs-apps/gpio_demo)

Payload Manager
[Payload Manager App](https://github.com/cfs-apps/pl_mgr)
[Payload Sim App](https://github.com/cfs-apps/pl_sim)
[Payload Sim Library](https://github.com/cfs-apps/pl_sim_lib)

Gonta01, 2024-05-16:
@open-stemware thank you a lot for you answers and provided links.

closed 2024-05-16 by Gonta01