nasa/cFS #767: Training guide apps and current cFS version compatibility.

Gonta01, 2024-05-17:
I've recently followed [CFS-101 TRAINING GUIDE](https://software.nasa.gov/software/MSC-26323-1) and successfully finished it.
Next I tried to repeat those steps with current version of cFS (main branch). I added [CI](https://github.com/nasa/CFS_CI) [TO](https://github.com/nasa/CFS_TO) [SCH](https://github.com/nasa/SCH) and [IO_LIB](https://github.com/nasa/CFS_IO_LIB).
Modified targets.cmake, so that instead of SET(cpu1_APPLIST ci_lab to_lab sch_lab) I use SET(cpu1_APPLIST io_lib ci to sch).
After this troubles began... 
First thing first after I tried to build project I got:
[image]
Then I found at path to/fsw/examples/README where it is written to use command ./setup.sh -m MISSIONNAME udp. After this and renaming CFS_TST_to_types.h into MISSIONNAME_to_types.h I got the next problem:
[image]
The problem is that I couldn't find CFE_SB_TLM_HDR_SIZE definition anywhere. Even in training guide Virtual box machine it isn't clear for me where it is defined...

Could someone give a hint what I'm missing or recommend some documentation on it. Could it be the problem of incompatibility of old apps with current cFS version?

spaceusername, 2024-05-17:
I've noticed that there is significant de-sync on some apps that are listed on the main cFS repo. The best guide that I could find for updating everything is here:
https://github.com/OpenSatKit/OpenSatKit/wiki/App-Migration-Notes-from-cFE-6.7-to-Caelum

Other than that I was able to find miscellaneous discussion threads and pull requests for other deviations. Unfortunately, I don't know if I retained the others, but this should be enough to hit most everything.

avan989, 2024-08-26:
Those macros have probably been updated/changed (in the cfe submodule). If you need to migrate some old apps to the current one, look at the sample_app as an example.

dzbaker, 2024-09-12:
Most up-to-date training can be found here: https://ntrs.nasa.gov/api/citations/20240000217/downloads/cFS_Training_Draco.pptx. Closing issue (please follow up in Discussions if there are more questions).

closed 2024-09-12 by dzbaker