nasa/cFS #770: cFS releases

Gonta01, 2024-05-28:
Could somebody explain cFS releases. What means [equuleus-rc1](https://github.com/nasa/cFS/releases/tag/equuleus-rc1), [draco-rc4](https://github.com/nasa/cFS/releases/tag/draco-rc4), [cFS Caelum Release Candidate 4](https://github.com/nasa/cFS/releases/tag/caelum-rc4), [cFS Bootes Release Candidate 2](https://github.com/nasa/cFS/releases/tag/bootes-rc2)? 
The latest release is [cFS Bundle - Aquila](https://github.com/nasa/cFS/releases/tag/v6.7.0a). Why so?

The main question is: if I want to use cFS for mission, what is better to choose now? Is it good idea to use just main branch, not releases?

ivanperez-keera, 2024-07-03:
I'm not a cFS dev and I'm not answering on behalf of the cFS team.

My understanding is that releases are in alphabetical order: https://github.com/nasa/cFS/tags

- `caelum` was released in 2021
- `draco` was released in 2022
- `equulus` was released in 2023

When the cFS team makes a release candidate (e.g., `draco-rc1`), they tag is as 'prerelease'. For example, in https://github.com/nasa/cFS/releases/tag/draco-rc4 there's a text that reads:

> Pre-release for stakeholder evaluation.

The way that versions are internally numbered is by `<last_major_release_candidate_name>+dev<incremental_version_numer>`. For example, `draco-rc4` is also `caelum-rc4+dev38` because it was not a final release, it was a pre-release.

When a release candidate requires no further changes and becomes final, they don't later come back to the releases tab and rename that to `draco-1.0.0`, or give it a separate tag name. Instead, they keep the tag `draco-rc5`, which is a bit unusual. Looking at the changelogs and searching for "baseline" can sometimes help identify what is happening.

Sublibraries (`cfe`, `osal`) are a bit different: if you look at the changelogs, they use `<last_major_release_version_number>+dev<incremental_version_numer>`. So, cFS itself is just a collection of libraries pinned to specific versions, with some configuration files, compilation scripts, and sample apps.

My recommendation is to never depend on a rolling HEAD (e.g., a development `main` branch that changes frequently): too much can change while a version is being developed and upgrading could delay your mission. Instead, use the latest release that is not tagged as pre-release. In this case, `draco-rc5`. I'd also recommend monitoring the `main` branch for any bug fixes identified that you may need to port over to your code.

RichLandau, 2024-07-03:
cFS team here. Thanks for helping to answer the question Ivan. The cFS core team has had some manpower issues over the last few years which we are trying to correct. These issues have led to release candidates being created, but there is a process for formal releases that has not happened in quite a bit. It's true that Aquila is the last formal release, but many rc's have been developed since then. They have been tested to the same level, they just haven't been formally released and are for stakeholder evaluation purposes. There is nothing to say you can't use one, you just need to test it with your system (which you need to do for formal releases anyway). We hope to release draco as an official release in the next few months and start having formal builds on a more regular cadence once again.

ivanperez-keera, 2024-07-04:
@RichLandau thanks for the clarification.

dzbaker, 2024-09-12:
Closing (follow up in Discussions if there are further questions).

closed 2024-09-12 by dzbaker