nasa/cFS #775: documentation about the pieces

schaefer01, 2024-05-31:
not so much a bug as a documentation request

I didn't know there were pieces until I found the soon-to-be-obsolete sourceforge link that pointed
   to the pieces:

https://github.com/nasa/SC 
https://github.com/nasa/SCH 
https://github.com/nasa/FM 
https://github.com/nasa/MM 
https://github.com/nasa/HK 
https://github.com/nasa/HS 
https://github.com/nasa/DS 
https://github.com/nasa/CF 
https://github.com/nasa/SBN 

could you put cFS under a different directory, for example, https://github.com/nasa/cFS_and_pieces/?
the reason is,  https://github.com/nasa/ contains a lot of stuff, so the cFS pieces are not easy to find

also, when I grab cFS through git are these parts included in the build of cFS?
and if so, then in what subdirectory can I find their source?

schaefer01, 2024-05-31:
I apologize, I realize now this should go into discussions

dzbaker, 2024-09-12:
Closing as cFS apps, libraries, and tools are listed in README.

closed 2024-09-12 by dzbaker