nasa/cFS #777: draco does not build

schaefer01, 2024-05-31:
hi,
  following the instructions in cFS_Training_Draco.pptx

-> the "make" returns errors and does not build
       there is no directory /home/rps/fsw/cFS/apps/to_lab/config
 
$ cp cfe/cmake/Makefile.sample Makefile
$ cp -r cfe/cmake/sample_defs sample_defs

$ make SIMULATION=native prep  
$ make
make --no-print-directory -C "build" mission-all
Scanning dependencies of target elf2cfetbl
[ 50%] Building C object tools/elf2cfetbl/CMakeFiles/elf2cfetbl.dir/elf2cfetbl.c.o
[100%] Linking C executable elf2cfetbl
[100%] Built target elf2cfetbl
[100%] Built target cfe-build-env
inside get_version for MISSION
inside get_version for core_api
inside get_version for core_private
inside get_version for ci_lab
inside get_version for to_lab
inside get_version for sch_lab
inside get_version for cfe_assert
inside get_version for sample_app
inside get_version for sample_lib
inside get_version for es
inside get_version for evs
inside get_version for fs
inside get_version for sb
inside get_version for tbl
inside get_version for time
inside get_version for osal
inside get_version for psp
inside get_version for msg
inside get_version for sbr
inside get_version for resourceid
inside get_version for config
[100%] Built target cfe-module-version
[100%] Built target mission-version
[100%] Built target mission-prebuild
Scanning dependencies of target tblobj_cpu1_sample_app.tble9d0d2cd826837dec30a2e776149f61929fd99ff
[  0%] Building C object apps/sample_app/CMakeFiles/tblobj_cpu1_sample_app.tble9d0d2cd826837dec30a2e776149f61929fd99ff.dir/fsw/tables/sample_app_tbl.c.o
[ 33%] Linking C static library libtblobj_cpu1_sample_app.tble9d0d2cd826837dec30a2e776149f61929fd99ff.a
[ 33%] Built target tblobj_cpu1_sample_app.tble9d0d2cd826837dec30a2e776149f61929fd99ff
[ 33%] Generating /home/rps/fsw/cFS/build/tables/cpu1_sample_app.tble9d0d2cd826837dec30a2e776149f61929fd99ff.sample_app_tbl.d
[ 33%] Built target generate_table_cpu1_sample_app_sample_app_tbl
Scanning dependencies of target tblobj_cpu1_to_lab.tblb058e9016025f9125d6a4462db60bbff759e3f18
[ 33%] Building C object apps/to_lab/CMakeFiles/tblobj_cpu1_to_lab.tblb058e9016025f9125d6a4462db60bbff759e3f18.dir/fsw/tables/to_lab_sub.c.o
In file included from /home/rps/fsw/cFS/apps/to_lab/fsw/tables/to_lab_sub.c:31:
/home/rps/fsw/cFS/build/native/default_cpu1/inc/to_lab_msgids.h:8:10: fatal error: /home/rps/fsw/cFS/apps/to_lab/config/default_to_lab_msgids.h: No such file or directory
    8 | #include "/home/rps/fsw/cFS/apps/to_lab/config/default_to_lab_msgids.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[8]: *** [apps/to_lab/CMakeFiles/tblobj_cpu1_to_lab.tblb058e9016025f9125d6a4462db60bbff759e3f18.dir/build.make:63: apps/to_lab/CMakeFiles/tblobj_cpu1_to_lab.tblb058e9016025f9125d6a4462db60bbff759e3f18.dir/fsw/tables/to_lab_sub.c.o] Error 1
make[7]: *** [CMakeFiles/Makefile2:1875: apps/to_lab/CMakeFiles/tblobj_cpu1_to_lab.tblb058e9016025f9125d6a4462db60bbff759e3f18.dir/all] Error 2
make[6]: *** [CMakeFiles/Makefile2:846: CMakeFiles/cfetables.dir/rule] Error 2
make[5]: *** [Makefile:188: cfetables] Error 2
make[4]: *** [CMakeFiles/native_default_cpu1-cfetables.dir/build.make:57: CMakeFiles/native_default_cpu1-cfetables] Error 2
make[3]: *** [CMakeFiles/Makefile2:213: CMakeFiles/native_default_cpu1-cfetables.dir/all] Error 2
make[2]: *** [CMakeFiles/Makefile2:303: CMakeFiles/mission-all.dir/rule] Error 2
make[1]: *** [Makefile:214: mission-all] Error 2
make: *** [Makefile:119: all] Error 2

jphickey, 2024-06-11:
Did you do a `git submodule init` and `git submodule update`?  What does `git status` say?

irowebbn, 2024-06-11:
I'm unable to replicate with the steps from the aforementioned [training slides](https://ntrs.nasa.gov/citations/20240000217) and I suspect this should be moved over to [discussions](https://github.com/nasa/cFS/discussions)

schaefer01, 2024-06-11:
my history shows these steps:

 1909  git checkout draco-rc5 
 1910  git submodule update --init

with no errors noted at that time

Is there a difference between

the single command:

   git submodule update --init

and the two commands:

  git submodule init 
  git submodule update

$ git status
HEAD detached at draco-rc5
nothing to commit, working tree clean

dzbaker, 2024-11-21:
CCB 21 November 2024: Have confirmed that draco is able to build; unable to recreate. Feel free to repost this as a discussion.

closed 2024-11-21 by dzbaker