nasa/cFS #785: Non Flight Apps

blueed9696, 2024-07-24:
I'm trying to build a RF communication app and trying to reference other app codes. However, I could find the apps in this code are non flight applications.

What makes them to become a non flight purpose? Is it because it doesn't have RF system?

avan989, 2024-08-20:
I think it is a liability thing. Every flight system is different and you would need to configure the app for it and make sure it is a fit for your need.

BBinLP, 2024-08-20:
It has been our experience that RF communications is very unique from one
mission to the next.  Whether it is due to the transceiver hardware,
multiple antennas, encryption, data rates, or how the hardware is attached
to the spacecraft bus.  Some efforts have been made recently to at least
separate the protocol processing (e.g. - CCSDS standards) from the hardware
interface but that is not always as simple as it sounds either because
special operational circumstances seem to always pop up.

dzbaker, 2024-09-12:
Closing issue. Please continue in discussions.

closed 2024-09-12 by dzbaker