nasa/cFS #811: Security.md: Grammatech CodeSonar link is 404

douglasheld2, 2024-12-11:
**Checklist (Please check before submitting)**

* [x ] I reviewed the [Contributing Guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md).
* [ x] I performed a cursory search to see if the bug report is relevant, not redundant, nor in conflict with other tickets.

**Describe the bug**
Security.md contains a dead link describing CodeSonar

**To Reproduce**
Steps to reproduce the behavior:
1. Browse to https://github.com/nasa/cFS/security#codesonar
2. Click on the link
4. See error 404 from Grammatech

**Expected behavior**
Information about CodeSonar

**System observed on:**
 - Apple MacBookPro15,1
 - OS: MacOS Sonoma 14.7.1
 - Google Chrome Version 131.0.6778.109 (Official Build) (x86_64)

**Additional context**
N/A

**Reporter Info**
Douglas Held

vinsdragonis, 2024-12-20:
I would like to take up this issue if it is still open.

Thanks.
@vinsdragonis

vinsdragonis, 2024-12-20:
I have made the changes in the PR: [CodeSonar URL correction](https://github.com/nasa/cFS/pull/813)

RichLandau, 2024-12-20:
@vinsdragonis certainly - please review the [Contributing guide](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md) and open a pull request. Appreciate the contribution!

commit d22f30ba (2024-12-30): Merge pull request #814 from vinsdragonis:fix-811-updated-security.md

Fix #811, Updated SECURITY.md to fix error 404 for CodeSonar link
closed 2024-12-30 by dzbaker