nasa/cFS #824: `CFE_SB_AllocateMessageBuffer()` fails in multi-threaded context

arya-pathak, 2025-02-23:
I am running cFS over Linux on an x86-64 machine. In the `main` function of my cFS app, I have created a thread through `pthread_create()`. Inside this thread, I read from a socket, and on receiving data I call another function that sends this data over the software bus. However, here the call to `CFE_SB_AllocateMessageBuffer()` always fails and returns `NULL`. On further inspection, the call to `CFE_ES_GetAppID()` is failing. This was never happening when the app was single-threaded. Any fixes for this?

CC: @rkshthrmsh

jphickey, 2025-02-24:
Don't use `pthread_create`.  Use  `CFE_ES_CreateChildTask` to create child tasks in your app.

arya-pathak, 2025-02-24:
This fixes the issue, thanks. Why does a newly created thread from native OS APIs lose context?

arya-pathak, 2025-02-25:
Could you help me understand the consequences of using native OS APIs instead of cFE/OSAL-specific ones? For operations tarring or copying files, I'm using `system("cp a b");` or `system("tar -xvf cpu1.tar.xz");` and these are behaving unreliably. Is there any way to do this through OSAL/cFE APIs? If not, is it dangerous to use Linux' `system()` function?