nasa/cFS #846: itc_synopsis_bridge.h missing

ddm3rr1tt, 2025-04-24:
git clone
git submodule

make prep
make

then t some point:

[100%] Built target amd64_nos3_default_cpu1-cfetables
/home/ex0atmospheric/nos3/components/syn/fsw/cfs/src/syn_app.c:14:10: fatal error: itc_synopsis_bridge.h: No such file or directory
   14 | #include "itc_synopsis_bridge.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Building C object apps/sample/fsw/cfs/CMakeFiles/sample.dir/home/ex0atmospheric/nos3/components/sample/fsw/shared/sample_device.c.o
make[7]: *** [apps/syn/fsw/cfs/CMakeFiles/syn.dir/build.make:76: apps/syn/fsw/cfs/CMakeFiles/syn.dir/src/syn_app.c.o] Error 1
make[6]: *** [CMakeFiles/Makefile2:5264: apps/syn/fsw/cfs/CMakeFiles/syn.dir/all] Error 2

ivanperez-keera, 2025-04-25:
Try to do:

`git submodule update --init --recursive`

before `make prep`

ivanperez-keera, 2025-05-05:
Would you like to submit a PR that fixes the doc as needed?

commit d2312a69 (2021-03-24): IC:2021-03-30, Add osal Fast Tracks

nasa/osal#878 - Fix #843, remove BIG/LITTLE bit order macros

nasa/osal#918 - Fix #846, Minor clean up and clarification in comments/namin